balena-os/leviathan #356: Check integrity of Images on upload

vipulgupta2048, 2020-09-09:
Got some feedback to work on improving testbot from the 3rd Sept Release Party 

Current workflow --> flaky uploads of OS image leads to image corruption --> An corrupted os image is then flashed and used in the device to flash and wait for it to come online and eventually fails --> Which leads to time taking flaky tests.

Integrating an OS image integrity check `beforeFlash` would really be time-saving. 
cc: @roman-mazur

roman-mazur, 2020-09-14:
Quick thoughts: we can modify the core to return a fingerprint of the image once the upload is done. If the client detects the mismatch, it retries the upload.

vipulgupta2048, 2020-09-14:
Exactly what I was thinking, that way we start another upload way faster. 
I will get started right on that :+1:

vipulgupta2048, 2021-02-18:
@rcooke-warwick We should try this again one more time as well :joy:

vipulgupta2048, 2021-07-15:
@rcooke-warwick Do we check for integrity now, the last time we looked into Zlibs?

vipulgupta2048, 2022-01-18:
No longer required, check current mechanism for upload #567

closed 2022-01-18 by vipulgupta2048