balena-os/leviathan #379: Fixing the timeouts when uploading an image from client to testbot

rcooke-warwick, 2021-04-07:
We are seeing many errors when integrating with jenkins, where the upload from the client->target testbot is timing out.

rcooke-warwick, 2021-04-07:
Recently I've seen the rig is being a lot more consistant, we increased the timeout and as far as I can tell, there haven't been as many timeouts since. Need to investigate further

klutchell, 2021-09-24:
We saw this failure when uploading suites as well on RPi4.
```
[2021-09-24T09:25:50.779Z][af82ff1-hup] Handling artifact: suite
[2021-09-24T09:25:50.779Z][af82ff1-hup] Calculating hash
[2021-09-24T09:25:50.788Z][af82ff1-hup] Sending to the testbot device, attempt 1...
...
[2021-09-24T09:45:50.797Z][af82ff1-hup] Error: Upload timed out
    at Timeout.setTimeout (/usr/app/lib/main.js:219:17)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
[2021-09-24T09:45:50.912Z][af82ff1-hup] WS connection is closed
```
https://11a1d0258e8782b188c3598b57c155c6.balena-devices.com/job/leviathan-raspberrypi4-64/473/console
https://www.flowdock.com/app/rulemotion/device-testing/threads/dogNHVh4_2idGgdcsE9ViWBZ38I

vipulgupta2048, 2022-01-18:
Doesn't happen all that much.

closed 2022-01-18 by vipulgupta2048