balena-os/leviathan #380: Make the programming interface simpler

rcooke-warwick, 2021-04-07:
At the moment, we have a set of methods in worker.js that allow you to interact with the DUT to

- flash
- power on/off
- set testbot network ap
- execute commands
- push locally

We need to simplify naming, or make it more clear what helpers are available, We could also add more helpers

rcooke-warwick, 2021-04-08:
Add a reboot and wait function:

Currently in about 5 OS tests, there is the same snippet of code that:

- creates a file in /tmp
- reboots the DUT via /sbin/reboot
- tries to ssh into the DUT and checks if that file is present in /tmp - if it isn't we can say the DUT has rebooted
- it should have built in retry/wait

vipulgupta2048, 2021-07-15:
Completed this one with #439, and a bigger goal for #436

closed 2021-07-15 by vipulgupta2048