balena-os/leviathan #389: Details button should link to actual leviathan job rather than leviathan-set-meta-balena-status

vipulgupta2048, 2021-04-21:
Need to investigate if we can improve UX by changing the link to the actual leviathan job that runs for each of the tests below. Currently they link to leviathan-set-meta-balena-status which confuses people and isn't helpful. 

[image]

vipulgupta2048, 2022-01-18:
This is the URL that Jenkins provides to GitHub on a per job basis. 
Can actually be solved now since we don't run jobs on meta-balena making set-meta-balena-status job useless 
@klutchell What do you think?

klutchell, 2022-01-18:
We still run generic jobs on meta-balena, and they still sometimes link to the status job for some reason.