balena-os/leviathan #481: Leviathan Suite for E2E testing DUT + Testbot setup for consistency across fleets

Bucknalla, 2021-08-25:
This should be a self-test style suite that allows the user to determine if their hardware setup matches that of the fleet the testbot + DUT is designed to function within. 

Discussed here -> https://www.flowdock.com/app/rulemotion/device-testing/threads/N1KToCgH9pPKHXkEpqiRGKiWCqO

rcooke-warwick, 2021-08-31:
along with this (or integrated with it), we can have a test suite that can be used to test the framework. An e2e test that checks the entire stack is working, e.g :+1: 

- leviathan e2e - 
		- client should find testbot
		- send an image to a testbot
		- flash testbot
		- run a command on DUT
		- archiver to get journal logs

vipulgupta2048, 2021-08-31:
@Bucknalla I am taking a inital crack at this, this week. Let me know if you want to hop in.

Bucknalla, 2021-09-01:
@vipulgupta2048 do you have a branch with this work on it? Might not have time to help but definitely curious to follow along

vipulgupta2048, 2021-09-06:
You got it, will post updates on FD and create a branch.

closed 2022-01-18 by bulldozer-balena[bot]