balena-os/leviathan #494: UnhandledPromiseRejectionWarning on rebooting the DUT

vipulgupta2048, 2021-09-22:
https://11a1d0258e8782b188c3598b57c155c6.balena-devices.com/job/leviathan-raspberrypi3/507/console

```
[2021-09-21T19:13:02.563Z][27718d4-os] DUT has rebooted & is back online
[2021-09-21T19:13:05.605Z][27718d4-os]         # Subtest: os.network.connectivity test
[2021-09-21T19:13:05.605Z][27718d4-os]             ok 1 - NetworkManager should be configured with uri: http://www.archlinux.org/check_network_status.txt
[2021-09-21T19:13:38.086Z][27718d4-os]             1..1
[2021-09-21T19:13:38.087Z][27718d4-os]         ok 4 - os.network.connectivity test # time=90724.69ms
[2021-09-21T19:13:38.087Z][27718d4-os] 
[2021-09-21T19:13:55.161Z][27718d4-os] Rebooting the DUT
[2021-09-21T19:14:01.442Z][27718d4-os] (node:4880) UnhandledPromiseRejectionWarning: Error: No response from server
    at Socket.<anonymous> (/usr/app/node_modules/ssh2/lib/client.js:313:15)
    at Socket.emit (events.js:198:13)
    at Socket.EventEmitter.emit (domain.js:448:20)
    at TCP._handle.close (net.js:607:12)
[2021-09-21T19:14:01.442Z][27718d4-os] (node:4880) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
[2021-09-21T19:14:01.442Z][27718d4-os] (node:4880) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

jakogut, 2022-05-20:
@vipulgupta2048 @klutchell This is still occurring, do we have any information to narrow this down at all?

```
leviathan-v2-template-client-1  | [2022-05-20T17:33:41.611Z][worker-os]     # Subtest: fsck.ext4 tests
leviathan-v2-template-client-1  | [2022-05-20T17:33:41.611Z][worker-os]         # Subtest: ext4 filesystems are checked on boot
leviathan-v2-template-client-1  |             ok 1 - Filesystem state for resin-rootA should be 'clean with errors'
leviathan-v2-template-client-1  | [2022-05-20T17:33:44.369Z][worker-os]             ok 2 - Filesystem state for resin-state should be 'clean with errors'
leviathan-v2-template-client-1  | [2022-05-20T17:33:44.425Z][worker-os]             ok 3 - Filesystem state for resin-rootB should be 'clean with errors'
leviathan-v2-template-client-1  | [2022-05-20T17:33:52.499Z][worker-os]             ok 4 - Filesystem state for resin-data should be 'clean with errors'
leviathan-v2-template-client-1  | [2022-05-20T17:33:52.499Z][worker-os] Rebooting the DUT
leviathan-v2-template-worker-1  | dnsmasq-dhcp: DHCPREQUEST(brd3qjewrt) 10.10.10.76 52:54:00:12:34:56 
leviathan-v2-template-worker-1  | dnsmasq-dhcp: DHCPACK(brd3qjewrt) 10.10.10.76 52:54:00:12:34:56 a7e6325
leviathan-v2-template-client-1  | [2022-05-20T17:35:46.671Z][worker-os] DUT has rebooted & is back online
leviathan-v2-template-client-1  | [2022-05-20T17:36:00.900Z][worker-os] Error: No response from server
```

https://jenkins.product-os.io/job/leviathan-v2-template/952/console

jakogut, 2022-05-20:
My normal tricks don't work, grepping for "No response from server" produces nothing

jakogut, 2022-05-20:
Actually, from the original message, this appears to be originating from the SSH client. It looks like our new message is less helpful than the old one.

klutchell, 2022-05-20:
It looks like it only tried for 15s to get an SSH connection?
```
leviathan-v2-template-client-1  | [2022-05-20T17:35:46.671Z][worker-os] DUT has rebooted & is back online
leviathan-v2-template-client-1  | [2022-05-20T17:36:00.900Z][worker-os] Error: No response from server
```

Even though `DUT has rebooted & is back online` was printed suggesting at least one SSH connection worked?