balena-os/leviathan #516: balenaOS in container worker?

rcooke-warwick, 2021-11-02:
@klutchell is it worth us making a balenaOS in a container worker:  https://github.com/balena-os/balenaos-in-container

It feels like this may be useful? (Spin up virtual device, run tests like the cloud test suite on it). I know there is overlap with the qemu worker though so not sure if this would be worth doing or not

klutchell, 2021-11-02:
I think this is definitely worth it, and less friction than a qemu worker. Both should exist IMO, and IIRC Alex M specifically mentioned testing basic interfaces against this.

rcooke-warwick, 2021-11-03:
@klutchell I'll get onto it, quite keen to get this up and running - do you know if the jenkins yocto jobs create a draft base image that can be used with this? In the repo I linked, they are pulling from here: https://hub.docker.com/r/resin/resinos/tags, which only contains released versions as far as I can tell

klutchell, 2021-11-04:
> do you know if the jenkins yocto jobs create a draft base image that can be used with this?

Yeah, the Jenkins jobs attach a `.docker` extension in addition to the images.
\https://jenkins.product-os.io/view/Yocto/job/yocto-raspberrypi3/1065/

It's in the format of a docker save, so just need to download it and `docker load balena-image.docker`.

jakogut, 2021-11-12:
I completely agree. One of the benefits of running containerized is that you don't need to dedicate memory to it, it shares memory with the host system. There's also less time spent with things like booting.