balena-os/leviathan #526: RPi3 OS test suite job froze on screen capture

vipulgupta2048, 2021-11-11:
https://jenkins.product-os.io/job/leviathan-raspberrypi3/713/

vipulgupta2048, 2021-11-11:
Aborted but couldn't investigate due to release party today. Will come back ot this.

vipulgupta2048, 2021-11-11:
```
11.11.21 06:25:39 (+0530)  core  
11.11.21 06:25:39 (+0530)  core  > @ start /usr/app
11.11.21 06:25:39 (+0530)  core  > node lib/main.js
11.11.21 06:25:39 (+0530)  core  
11.11.21 06:25:42 (+0530)  core  Listening on port 80
11.11.21 06:27:28 (+0530)  worker  Start network setup
11.11.21 06:27:28 (+0530)  worker  Adding wireless connection...
11.11.21 06:27:28 (+0530)  worker  Wireless AP; SSID: oepha8sb IFACE: wlan0
11.11.21 06:27:28 (+0530)  worker  Network setup completed
11.11.21 06:29:42 (+0530)  worker  Powering off DUT...
11.11.21 06:29:42 (+0530)  worker  Start flashing...
11.11.21 06:33:19 (+0530)  worker  Flashing completed.
11.11.21 06:33:19 (+0530)  worker  Trying to power on DUT...
11.11.21 06:33:21 (+0530)  worker  Vout= 4.99375
11.11.21 11:33:19 (+0530)  worker  Performing teardown...
11.11.21 11:33:25 (+0530)  core  Suite exit code is: undefined
11.11.21 11:33:51 (+0530)  core  events.js:174
11.11.21 11:33:51 (+0530)  core        throw er; // Unhandled 'error' event
11.11.21 11:33:51 (+0530)  core        ^
11.11.21 11:33:51 (+0530)  core  
11.11.21 11:33:51 (+0530)  core  Error [ERR_STREAM_WRITE_AFTER_END]: write after end
11.11.21 11:33:51 (+0530)  core      at writeAfterEnd (_stream_writable.js:248:12)
11.11.21 11:33:51 (+0530)  core      at WriteStream.Writable.write (_stream_writable.js:296:5)
11.11.21 11:33:51 (+0530)  core      at ChildProcess.msgHandler (/usr/app/lib/main.js:172:14)
11.11.21 11:33:51 (+0530)  core      at ChildProcess.emit (events.js:198:13)
11.11.21 11:33:51 (+0530)  core      at ChildProcess.EventEmitter.emit (domain.js:448:20)
11.11.21 11:33:51 (+0530)  core      at emit (internal/child_process.js:832:12)
11.11.21 11:33:51 (+0530)  core      at process._tickCallback (internal/process/next_tick.js:63:19)
11.11.21 11:33:51 (+0530)  core  Emitted 'error' event at:
11.11.21 11:33:51 (+0530)  core      at errorOrDestroy (internal/streams/destroy.js:107:12)
11.11.21 11:33:51 (+0530)  core      at writeAfterEnd (_stream_writable.js:250:3)
11.11.21 11:33:51 (+0530)  core      at WriteStream.Writable.write (_stream_writable.js:296:5)
11.11.21 11:33:51 (+0530)  core      [... lines matching original stack trace ...]
11.11.21 11:33:51 (+0530)  core      at process._tickCallback (internal/process/next_tick.js:63:19)
11.11.21 11:33:51 (+0530)  core  npm ERR! code ELIFECYCLE
11.11.21 11:33:51 (+0530)  core  npm ERR! errno 1
11.11.21 11:33:51 (+0530)  core  npm ERR! @ start: `node lib/main.js`
11.11.21 11:33:51 (+0530)  core  npm ERR! Exit status 1
11.11.21 11:33:51 (+0530)  core  npm ERR! 
11.11.21 11:33:51 (+0530)  core  npm ERR! Failed at the @ start script.
11.11.21 11:33:51 (+0530)  core  npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
11.11.21 11:33:51 (+0530)  core  
11.11.21 11:33:51 (+0530)  core  npm ERR! A complete log of this run can be found in:
11.11.21 11:33:51 (+0530)  core  npm ERR!     /root/.npm/_logs/2021-11-11T06_03_51_839Z-debug.log
11.11.21 11:33:53 (+0530) Service exited 'core sha256:82b930393c8e34eed360263a2393e89fcee2b8547f653a97fe85d34d58405f2e'
11.11.21 11:33:52 (+0530)  core  
```

klutchell, 2021-11-11:
My device didn't have the HDMI to USB interface, but we should have caught this error instead of hanging.