balena-os/leviathan #567: Rework Upload Mechanism

vipulgupta2048, 2021-12-09:
Upload atm in Leviathan is complex for no good reason. 

## Context

The flow is client pings `/start` endpoint of core, core responds with requests for items (with a hash for security purposes...) and it their it asks for a os.img file, client receives this request through their websocket connection.

Client responds by looking for the path of the image that it has and processes it along with that same hash to verify that the files have come from the right place. It does the same for the other artifacts. Core calculates the hash of the file, matches the hash with the files it already has and if they don't match, then it starts the upload of the files and receives them. 

**Why is the core one asking for files, reasons:**
 
1. Security concern - Previous contributors wanted to avoid anyone sending anything to the testbots. With this current method, the core generates these tokens, and what the client sends must have the same token. (This is broken as even though the tokens need to be the same, you can send a file with whatever type or contents. No checks are done.)
2. It knows where to put them artifacts and how to process
3. Caching

## Rework

Client sends a set of artifacts, core calculates hash with what artifacts it has, decides caching, receives files, processes files. The core should only be accepting a set defined list of artifacts (example: a config.json file, a suite, and an image file). As more test suites want to send more artifacts to the core, we can rethink our decision to extend the set of artifacts that the core will be receiving. 

Vipul will try a PoC.

vipulgupta2048, 2022-01-05:
## Investigation into upload mech

> issue https://github.com/balena-os/leviathan/issues/567

1. Client starts the WS connection to `ws://${this.uri.host}/start`
2. Core handles the /start endpoint and returns a list of artifacts it wants and a security token
3. Client receives said list from code and passes it off to the wsMessageHandler
4. According to the what the artifact ID is, the wsMessageHandler will decide what to do with it. 
5. wsMessageHandler will pass on the artifact process to handleArtifact function where they are processed, gzipped, hash calculated and later uploaded to the core on the /upload endpoint with the security token. 
6. Here's how upload works, 

- Client sends a message to core with security token, file name, file type, file path, file hash.
- Core accepts, and uses the path sent to check for if any files exist on that path
- If FOUND, core uses the file type to calculate hash of the file that exists on the core side. 
- IF the HASH matches, then the cache is used and the upload doesn't take place. 
- IF the hash doesn't match then the file is deleted? and the upload process starts
- Throughout this process, core has an intricate retry process where if the upload fails. Then core lets the client know of exactly how many attempts are remaining for that particular file. We allow 3 attempts to upload. Core is the one that keeps count and retries the upload process on the client side using it's websocket /start connection. 
- When the upload process concludes, the core sends back the done message and the upload is complete. 
- After that the core starts the tests and the log message is sent back which then is used on client side to get logs from the tests running on the core and the test process begins. 

Questions 

- Should we keep caching in favor of a configurable upload or not switch?
  - If you use a switch like that, then what is the use of the caching. 
  - Pros: Easier code to read and less time for calculating the hash. 
  - Cons:

- We should keep security tokens around. 

- Check hash creation for directories, and remove duplicate code

- Why can't websockets handle image uploads and why we need a HTTP server.

 - Figure our a sane retry mechanism

- Clean out interactive code

jakogut, 2022-02-03:
> Should we keep caching in favor of a configurable upload or not switch?

I might need some clarification on this one

> We should keep security tokens around.

This seems nonsensical to me, just don't allow unauthorized users to upload or use testbots. If we transfer data/files over authorized SSH, the protocol itself checks and guarantees the integrity of the data.

> Why can't websockets handle image uploads and why we need a HTTP server.

The WS connection seems redundant/reinventing the wheel. Websockets are useful when you're communicating with browsers, which don't have native sockets. Between two node applications, there's no benefit over a native socket. Indeed, inventing a new protocol to go over a socket, whether native or web variety, in this context, is probably reinventing the wheel. SSH already does secure/verified/compressed transfer of text and binary data between two hosts.

vipulgupta2048, 2022-02-08:
Solution: Rsync with SSH

**Specification**: https://docs.google.com/document/d/160MKkqVHLRUEqUszCUwc4xSzF11fpR3mhdG3Orq5Tr8/edit?usp=sharing
**FD**: https://www.flowdock.com/app/rulemotion/p-testbot/threads/mo408W6ORf9ykP8JBKiGmBHz8nd

vipulgupta2048, 2022-04-20:
After recent discussions and moving core off the testbot. It becomes clear how the upload mechanism looks like. 

1. It doesn't exist for the most part. 

With #729, volumes will act as a way for core to have direct access to all artifacts needed to run tests. The only uploads that will be there would be the streaming of artifacts to the worker in order to flash the image. 

In a chance where we intend to send artifacts to the worker, we intend to create a Rsync tunnel for the back and forth of artifacts between core (testrunner) and worker

vipulgupta2048, 2023-02-14:
With the recent file upload limit imposed on `balena-devices.com` URLs, the current flashing as well as upload mechanism will have to change considerably and hence making the rsync discussion moot.

closed 2023-02-14 by vipulgupta2048