balena-os/leviathan #608: Update etcher-sdk to latest version

acostach, 2022-01-14:
We'll need etcher-sdk to be updated to the latest revision so we can use the node-raspberrypi-usbboot [changes](https://github.com/balena-io-modules/node-raspberrypi-usbboot/pull/41) for a custom Pi with emmc. I've attempted to do this in https://github.com/balena-os/leviathan/pull/550 however it did not work. Perhaps we can add this item to the todo list @vipulgupta2048 @rcooke-warwick

rcooke-warwick, 2022-01-18:
Hey @acostach I can help take a look at this as its blocking adding more devices - 

Is the bump to the etcher sdk for use in flashing the device? If so, then I'm not sure it actually needs to be bumped in this repo. The lower level code for flashing devices is in https://github.com/balena-io-hardware/testbotsdk

rcooke-warwick, 2022-01-18:
Had a look, and it looks like etcher sdk is used in the following places in this repo:

This function:
https://github.com/balena-os/leviathan/blob/91bd4cbc50d61db8a1de61e07e755092edc07136/worker/lib/helpers/index.ts#L28 
Which it turns out is no longer used anywhere.

Here:
https://github.com/balena-os/leviathan/blob/91bd4cbc50d61db8a1de61e07e755092edc07136/worker/lib/index.ts#L256
Which no longer works/is not longer used as far as I can see

Here:
https://github.com/balena-os/leviathan/blob/91bd4cbc50d61db8a1de61e07e755092edc07136/worker/lib/workers/qemu.ts#L133
Which is only used for qemu. 

So I think if we want to flash the Pi with emmc, we should bump it in the testbotsdk repo.

acostach, 2022-01-19:
I see, thanks for clarifying Ryan! indeed, it's for flashing the flashing the custom pi

acostach, 2022-01-31:
Closing for now, will re-open if necessary.

closed 2022-01-31 by acostach