balena-os/leviathan #610: Solve the inital fail that leads to false positive

vipulgupta2048, 2022-01-17:
https://github.com/balena-os/meta-balena/pull/2478 tests are leading to a false positive condition that can be resolved by outputting a error message here. Creating an issue so I don't forget. 

https://github.com/balena-os/leviathan/blob/810434b46e3d25ba16c074e97d15c33c1bcce49b/core/lib/main.js#L168-L178

rcooke-warwick, 2022-01-19:
I think https://github.com/balena-os/leviathan/pull/463 removes the false positive iirc

vipulgupta2048, 2022-02-03:
The suite Subprocess PR will make these handlers obsolete and hence removes the chance of a false positive occurring again.

jakogut, 2022-02-03:
@vipulgupta2048 I'm not sure it does, the code should work basically identically to how it does now.

closed 2022-04-25 by bulldozer-balena[bot]

vipulgupta2048, 2022-04-26:
Should have removed this link after @jakogut's comment. This problem still is present.