balena-os/leviathan #618: Improve docs for QEMU worker

lmbarros, 2022-01-20:
Maybe the introduction of QEMU workers is a feature large enough to justify restructuring the docs a bit more deeply. For example, *maybe* it would be worth to have two separate "quick starts", one for QEMU, another one for testbot.

Anyway, here's a list of more specific improvements I can think of:

* [ ] The `README` says that we shall run `make local` to run tests using a QEMU worker. This should be `make local-test` now, AFAIU. Also, should probably be mentioned in the quick start, too (or instead).
* [ ] The quick start guide says that "as a prerequisite for this guide, you will have to prepare your testbot". This is no longer true with QEMU workers.
* [ ] Would be very nice to have a `config.json` example for QEMU that just works out of the box (maybe using the e2e tests).
* [ ] Document how `networkWired` and `networkWireless` interact with QEMU workers.

jakogut, 2022-02-03:
For the last part, the `networkWired` and `networkWireless` don't interact with the QEMU worker at all.

vipulgupta2048, 2022-02-21:
I agree

closed 2022-02-28 by bulldozer-balena[bot]