balena-os/leviathan #738: Docs: Lean More Doc Links are Private

keenanjohnson, 2022-04-25:
Hello!

I tried to follow the Google drive links in [documentation/learn-more.md](https://github.com/balena-os/leviathan/blob/master/documentation/learn-more.md), but each of the google drive links was listed as private, and I wasn't able to access them.

keenanjohnson, 2022-04-26:
Oh I see. I was just reading the repo and was curious and thought this to be a bug more than an intended idea.

vipulgupta2048, 2022-04-27:
Right on, @keenanjohnson I will update this issue as we release more information about the product. For now, if you like to take Leviathan for a spin, you can follow this guide: https://balena-os.github.io/leviathan/pages/Getting-Started/quickstart/quickstart-qemu.html

Let us know how it goes. Closing this issue.

closed 2022-04-27 by vipulgupta2048