balena-os/leviathan #754: Client websocket connection doesn't handle ECONNRESET, resulting in an unhandled promise rejection

jakogut, 2022-05-23:
```
leviathan-v2-template2-client-1  | [2022-05-21T16:14:42.398Z][worker-hup] Uploading - 61%
leviathan-v2-template2-client-1  | [2022-05-21T16:14:45.985Z][worker-hup] Uploading - 71%
leviathan-v2-template2-client-1  | [2022-05-21T16:14:49.372Z][worker-hup] Uploading - 81%
leviathan-v2-template2-client-1  | [2022-05-21T16:14:50.703Z][worker-hup] (node:40) UnhandledPromiseRejectionWarning: Error: [WSHandlerError] Error: read ECONNRESET
leviathan-v2-template2-client-1  |     at Client.<anonymous> (/usr/src/app/build/lib/index.js:525:47)
leviathan-v2-template2-client-1  |     at step (/usr/src/app/build/lib/index.js:45:23)
leviathan-v2-template2-client-1  |     at Object.throw (/usr/src/app/build/lib/index.js:26:53)
leviathan-v2-template2-client-1  |     at rejected (/usr/src/app/build/lib/index.js:18:65)
leviathan-v2-template2-client-1  |     at processTicksAndRejections (internal/process/task_queues.js:97:5)
leviathan-v2-template2-client-1  | [2022-05-21T16:14:50.703Z][worker-hup] (node:40) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
leviathan-v2-template2-client-1  | (node:40) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
leviathan-v2-template2-core-1    | Error detected: Error [ERR_STREAM_PREMATURE_CLOSE]: Premature close
leviathan-v2-template2-client-1  | [2022-05-21T16:14:50.706Z][worker-hup] WS connection is closed
leviathan-v2-template2-client-1  | [2022-05-21T16:14:50.707Z][worker-hup] Child 40 got an error:
leviathan-v2-template2-client-1  | [2022-05-21T16:14:50.707Z][worker-hup] Error: read ECONNRESET
leviathan-v2-template2-client-1  |     at TCP.onStreamRead (internal/stream_base_commons.js:209:20)
leviathan-v2-template2-client-1  | [2022-05-21T16:14:50.707Z][worker-hup] Exiting client process...
leviathan-v2-template2-client-1  | Downloading DUT serial log with http://worker/dut/serial
leviathan-v2-template2-client-1  | Downloading artifacts
leviathan-v2-template2-client-1  | Downloading test summary with http://core:2000/reports/test-summary.json
leviathan-v2-template2-client-1  | Exiting with 1, client = 0, children: 0 (success),0 (success),1 (error)
leviathan-v2-template2-client-1 exited with code 1
Aborting on container exit...
```

alexgg, 2022-05-28:
Seen in [genericx86-64-ext#1097](https://jenkins.product-os.io/job/leviathan-v2-template/1097/)

klutchell, 2022-06-14:
@rcooke-warwick do you think this is addressed by https://github.com/balena-os/leviathan/pull/775 ?

klutchell, 2022-06-30:
This made things worse, abandoning this PR.