balena-os/leviathan #782: SyntaxError: Unexpected token < in JSON at position 0

klutchell, 2022-06-15:
All our tests end with this error, I suspect related to our test report generation
```
client_1  | Exiting with 0, client = 0, children: 0 (success)
client_1  | undefined:1
client_1  | <!DOCTYPE html>
client_1  | ^
client_1  | 
client_1  | SyntaxError: Unexpected token < in JSON at position 0
client_1  |     at JSON.parse (<anonymous>)
client_1  |     at /usr/src/app/build/bin/multi-client.js:286:49
client_1  |     at Array.forEach (<anonymous>)
client_1  |     at process.<anonymous> (/usr/src/app/build/bin/multi-client.js:284:53)
client_1  |     at process.emit (events.js:314:20)
```

vipulgupta2048, 2022-06-16:
@klutchell Can i get an example job for this?

klutchell, 2022-06-16:
Apparently not, I'm having trouble finding one on Jenkins today. I thought it happened on every build but I will keep an eye out.

It happens locally for me.

vipulgupta2048, 2022-06-17:
Closing. The issue was `test-summary.json` files from other tests that have had errors in the past inside them such as "Public device URL errors" which can lead to issues in parsing and creation of the summary. Once these reports are deleted then the summary generator can work on clean files from that point onwards.

closed 2022-06-17 by vipulgupta2048