balena-os/leviathan #803: Using `test.resolves` will throw on failure and prevent collection of journal logs

klutchell, 2022-07-13:
Either we are using this function incorrectly in the meta-balena suite, or something is broken in our tap.
https://node-tap.org/docs/api/asserts/#tresolvespromise--fn-message-extra

Example of our usage in meta-balena:
https://github.com/balena-os/meta-balena/blob/f2690121369a4481e735cfa79b0ae6da034009f1/tests/suites/os/tests/chrony/index.js#L84-L95

klutchell, 2022-07-13:
```
leviathan-client-1  | [2022-07-13T17:38:33.989Z][worker-hup]         not ok 1 - Should catch rejection
leviathan-client-1  | [2022-07-13T17:38:33.989Z][worker-hup]           ---
leviathan-client-1  | [2022-07-13T17:38:33.990Z][worker-hup]           at:
leviathan-client-1  | [2022-07-13T17:38:33.990Z][worker-hup]             line: 50
leviathan-client-1  | [2022-07-13T17:38:33.990Z][worker-hup]             column: 14
leviathan-client-1  | [2022-07-13T17:38:33.990Z][worker-hup]             file: /data/suite/tests/rollback-altboot.js
leviathan-client-1  | [2022-07-13T17:38:33.990Z][worker-hup]             type: Proxy
leviathan-client-1  | [2022-07-13T17:38:33.990Z][worker-hup]             function: run
leviathan-client-1  |           found:
leviathan-client-1  | [2022-07-13T17:38:33.990Z][worker-hup]             !error
leviathan-client-1  | [2022-07-13T17:38:33.991Z][worker-hup]             name: Error
leviathan-client-1  |             message: Fake error
leviathan-client-1  | [2022-07-13T17:38:33.991Z][worker-hup]             stack: |-
leviathan-client-1  | [2022-07-13T17:38:33.991Z][worker-hup]               Error: Fake error
leviathan-client-1  | [2022-07-13T17:38:33.991Z][worker-hup]                   at Proxy.run (/data/suite/tests/rollback-altboot.js:51:28)
leviathan-client-1  |           source: |
leviathan-client-1  | [2022-07-13T17:38:33.991Z][worker-hup]                            // 0 means file exists, 1 means file does not exist
leviathan-client-1  | [2022-07-13T17:38:33.991Z][worker-hup]                            await test.resolves(
leviathan-client-1  | [2022-07-13T17:38:33.991Z][worker-hup]             -------------^
leviathan-client-1  |                         Promise.reject(new Error("Fake error")),
leviathan-client-1  | [2022-07-13T17:38:33.992Z][worker-hup]                                    'Should catch rejection'
leviathan-client-1  | [2022-07-13T17:38:33.992Z][worker-hup]           ...
leviathan-client-1  | [2022-07-13T17:38:34.000Z][worker-hup] 
leviathan-client-1  | [2022-07-13T17:38:34.000Z][worker-hup]         Bail out! Should catch rejection
```

vipulgupta2048, 2022-07-18:
Looking at the docs + our code, it seems test.resolves expects promises to be handled in both events of resolution/rejection. From first looks of the example, it seems we should be catching the promise rejection too? 

`test.resolves` probably doesn't have any error catching functionality inbuilt to not hide errors.

klutchell, 2022-07-21:
Fair enough, thanks @vipulgupta2048 !

closed 2022-07-21 by klutchell