balena-os/leviathan #806: executeCommandOverSSH: connection errors like "Bad packet length" can halt the test instance

klutchell, 2022-07-15:
This is 100% reproducible with a test I have locally that for some reason always returns `Bad packet length: 1397966893` from the `node-ssh` client connection.

As expected, the SSH command is retried, but the test instance in the suite is no longer printing test results or test comments to the logs.

This example (which for some reason returns the connection error) will leave `test` in a bad state.
```node
test.is(
	await this.worker.executeCommandInHostOS(
		`balena volume create hello-world`,
		this.link,
	),
	'hello-world',
	'Should create hello-world volume'
);
```
```
leviathan-client-1  | [2022-07-15T02:45:43.253Z][worker-hup] getting ip of worker
leviathan-client-1  | [2022-07-15T02:45:43.275Z][worker-hup] ip route add 10.10.10.76 via 172.21.0.2
leviathan-client-1  | [2022-07-15T02:45:43.281Z][worker-hup]             # Waiting for DUT to be reachable
leviathan-client-1  | [2022-07-15T02:45:44.358Z][worker-hup]             # DUT flashed
leviathan-client-1  | [2022-07-15T02:45:45.485Z][worker-hup]             # checking for under-voltage reports in kernel logs...
leviathan-client-1  | [2022-07-15T02:45:46.445Z][worker-hup]             # ok - No under-voltage reports in the kernel logs
leviathan-client-1  | [2022-07-15T02:45:51.826Z][worker-hup] Connection err: Bad packet length: 1397966893
leviathan-client-1  | [2022-07-15T02:45:56.639Z][worker-hup] Connection err: Bad packet length: 1397966893
leviathan-client-1  | [2022-07-15T02:46:04.139Z][worker-hup]             ok 1 - Should create hello-world volume
# missing test.comment(s) here
leviathan-client-1  | [2022-07-15T02:46:04.139Z][worker-hup] image-nccf6vah
# missing test.comment(s) here
leviathan-client-1  | [2022-07-15T02:46:41.326Z][worker-hup] /mnt/data/resin-data/image-nccf6vah
# missing test.results(s) here
```

I've also seen this happen on Jenkins with other tests, but it is more difficult to reproduce outside of this particular test & qemu device combination.

klutchell, 2022-07-15:
Other examples, these are all with generic-aarch64
```
jenkins-leviathan-v2-template-2900-client-1  | [2022-07-15T02:05:37.668Z][worker-cloud]         # Subtest: SSH authentication in production mode
jenkins-leviathan-v2-template-2900-client-1  |             # Setting production mode in config.json...
jenkins-leviathan-v2-template-2900-worker-1  | dnsmasq-dhcp: DHCPREQUEST(brtwi0er5f) 10.10.10.76 52:54:00:12:34:56 
jenkins-leviathan-v2-template-2900-worker-1  | dnsmasq-dhcp: DHCPACK(brtwi0er5f) 10.10.10.76 52:54:00:12:34:56 ec61507
jenkins-leviathan-v2-template-2900-worker-1  | dnsmasq-dhcp: DHCPREQUEST(brtwi0er5f) 10.10.10.76 52:54:00:12:34:56 
jenkins-leviathan-v2-template-2900-worker-1  | dnsmasq-dhcp: DHCPACK(brtwi0er5f) 10.10.10.76 52:54:00:12:34:56 ec61507
```
```
jenkins-leviathan-v2-template-2903-client-1  | [2022-07-15T01:37:21.217Z][worker-os]             # Reloading udev rules...
jenkins-leviathan-v2-template-2903-worker-1  | dnsmasq-dhcp: DHCPREQUEST(brw9vtxcoo) 10.10.10.76 52:54:00:12:34:56 
jenkins-leviathan-v2-template-2903-worker-1  | dnsmasq-dhcp: DHCPACK(brw9vtxcoo) 10.10.10.76 52:54:00:12:34:56 7681f82
jenkins-leviathan-v2-template-2903-worker-1  | dnsmasq-dhcp: DHCPREQUEST(brw9vtxcoo) 10.10.10.76 52:54:00:12:34:56 
jenkins-leviathan-v2-template-2903-worker-1  | dnsmasq-dhcp: DHCPACK(brw9vtxcoo) 10.10.10.76 52:54:00:12:34:56 7681f82
jenkins-leviathan-v2-template-2903-worker-1  | dnsmasq-dhcp: DHCPREQUEST(brw9vtxcoo) 10.10.10.76 52:54:00:12:34:56 
jenkins-leviathan-v2-template-2903-worker-1  | dnsmasq-dhcp: DHCPACK(brw9vtxcoo) 10.10.10.76 52:54:00:12:34:56 7681f82
```
The trailing DHCP requests show that the worker is still alive, but the test suite is stuck.

klutchell, 2022-07-15:
Captured this stack when receiving "Bad packet length"
```
leviathan-client-1  | 
leviathan-client-1  | Error: 
leviathan-client-1  |     at AESGCMDecipherBinding.decrypt (/usr/app/node_modules/ssh2/lib/protocol/crypto.js:1027:17)
leviathan-client-1  |     at Protocol.parsePacket [as _parse] (/usr/app/node_modules/ssh2/lib/protocol/Protocol.js:1994:25)
leviathan-client-1  |     at Protocol.parse (/usr/app/node_modules/ssh2/lib/protocol/Protocol.js:293:16)
leviathan-client-1  |     at Socket.<anonymous> (/usr/app/node_modules/ssh2/lib/client.js:713:21)
leviathan-client-1  |     at Socket.emit (events.js:400:28)
leviathan-client-1  |     at addChunk (internal/streams/readable.js:293:12)
leviathan-client-1  |     at readableAddChunk (internal/streams/readable.js:267:9)
leviathan-client-1  |     at Socket.Readable.push (internal/streams/readable.js:206:10)
leviathan-client-1  |     at TCP.onStreamRead (internal/stream_base_commons.js:188:23)
leviathan-client-1  |     at TCP.callbackTrampoline (internal/async_hooks.js:130:17)
leviathan-client-1  | Emitted 'error' event on Client instance at:
leviathan-client-1  |     at Socket.<anonymous> (/usr/app/node_modules/ssh2/lib/client.js:715:20)
leviathan-client-1  |     at Socket.emit (events.js:400:28)
leviathan-client-1  |     [... lines matching original stack trace ...]
leviathan-client-1  |     at TCP.callbackTrampoline (internal/async_hooks.js:130:17) {
leviathan-client-1  |   tapCaught: 'uncaughtException',
leviathan-client-1  |   test: 'HUP from previous release'
leviathan-client-1  | }
```

rcooke-warwick, 2022-07-15:
@klutchell is it always with this command

```
await this.worker.executeCommandInHostOS(
	`balena volume create hello-world`,
	this.link,
)
```

klutchell, 2022-07-15:
@rcooke-warwick look at all of the recent `qemu-generic-aarch64` jobs on Jenkins. They all seem to hang sometime after the SSH connection returns "Bad packet length".

This device used to work pretty reliably so something changed in the last week or two. I tried reverting some of the recent leviathan changes but it didn't help or made things worse. I hope it's not a dependency in the Alpine repos that changed, cause that would be very difficult to track down.

klutchell, 2022-07-15:
One the 3rd attempt this specific function will resolve
```
[2022-07-15T14:19:11.755Z][worker-hup] Outbound: Sending CHANNEL_REQUEST (r:0, exec: balena volume create hello-world )
[2022-07-15T14:19:11.755Z][worker-hup] Inbound: REQUEST_SUCCESS
[2022-07-15T14:19:11.761Z][worker-hup] Inbound: CHANNEL_WINDOW_ADJUST (r:0, 2097152)
[2022-07-15T14:19:11.761Z][worker-hup] Inbound: CHANNEL_SUCCESS (r:0)
Outbound: Sending CHANNEL_EOF (r:0)
[2022-07-15T14:19:19.086Z][worker-hup] Inbound: CHANNEL_DATA (r:0, 12)
[2022-07-15T14:19:19.100Z][worker-hup] Inbound: CHANNEL_EOF (r:0)
[2022-07-15T14:19:19.100Z][worker-hup] Outbound: Sending CHANNEL_CLOSE (r:0)
[2022-07-15T14:19:19.102Z][worker-hup] Inbound: CHANNEL_REQUEST (r:0, exit-status: 0)
[2022-07-15T14:19:19.103Z][worker-hup] Inbound: CHANNEL_CLOSE (r:0)
[2022-07-15T14:19:19.103Z][worker-hup] Outbound: Sending DISCONNECT (11)
[2022-07-15T14:19:19.105Z][worker-hup]             ok 1 - Should create hello-world volume
```
But the first two attempts both look like this
```
[2022-07-15T14:19:00.881Z][worker-hup] Outbound: Sending CHANNEL_REQUEST (r:0, exec: balena volume create hello-world )
[2022-07-15T14:19:00.882Z][worker-hup] Inbound: REQUEST_SUCCESS
[2022-07-15T14:19:00.892Z][worker-hup] Inbound: CHANNEL_WINDOW_ADJUST (r:0, 2097152)
[2022-07-15T14:19:00.892Z][worker-hup] Inbound: CHANNEL_SUCCESS (r:0)
Outbound: Sending CHANNEL_EOF (r:0)
[2022-07-15T14:19:05.630Z][worker-hup] Connection error: Bad packet length: 1397966893
```
```
[2022-07-15T14:19:06.462Z][worker-hup] Outbound: Sending CHANNEL_REQUEST (r:0, exec: balena volume create hello-world )
[2022-07-15T14:19:06.462Z][worker-hup] Inbound: REQUEST_SUCCESS
[2022-07-15T14:19:06.470Z][worker-hup] Inbound: CHANNEL_WINDOW_ADJUST (r:0, 2097152)
[2022-07-15T14:19:06.471Z][worker-hup] Inbound: CHANNEL_SUCCESS (r:0)
Outbound: Sending CHANNEL_EOF (r:0)
[2022-07-15T14:19:10.433Z][worker-hup] Socket ended
[2022-07-15T14:19:10.434Z][worker-hup] Socket closed
[2022-07-15T14:19:10.434Z][worker-hup] Outbound: Sending CHANNEL_CLOSE (r:0)
[2022-07-15T14:19:10.847Z][worker-hup] Connection error: Bad packet length: 1397966893
```

klutchell, 2022-07-15:
I've done some more testing and confirmed it does not matter what the promise is, or whether it returns on the first try.

For whatever reason, the first promise inside a tap assert will always break the tap logs for the rest of the suite. Maybe it's something specific to the HUP suite but I've now ruled out the device type as I'm not even calling an SSH function.

```node
module.exports = {
	title: 'Smoke tests',
	tests: [
		{
			title: 'HUP from previous release',
			run: async function (test) {
				await this.hup.initDUT(this, test, this.link);

				const activePartition = await this.worker.executeCommandInHostOS(
					`findmnt --noheadings --canonicalize --output SOURCE /mnt/sysroot/active`,
					this.link,
				);

				// Check for under-voltage before HUP, in the old OS
				await this.hup.checkUnderVoltage(this, test); // works

				test.ok(true, 'Should be truthy', { diagnostic: true }); // works

				test.comment("After should be truthy"); // works

				test.resolves(async () => {
					return Promise.resolve();
				}, 'Should resolve'); // resolves

				test.comment("After should resolve"); // does not print
```

klutchell, 2022-07-18:
Closing this for now until I can document the weirdness around promises inside test assertions and open a new bug.

The `Bad packet length` seems to be a red herring and any promise inside a test assert will result in odd test output.