balena-os/leviathan #811: Job passes with Success when Test Device Unsupported

vipulgupta2048, 2022-07-25:
Jenkins: https://jenkins.product-os.io/job/leviathan-v2-template/3208/console

```
jenkins-leviathan-v2-template-3208-client-1  | [2022-07-24T19:43:35.397Z][90b38fd-cloud] Uploading - 100%
jenkins-leviathan-v2-template-3208-client-1  | [2022-07-24T19:43:37.681Z][90b38fd-cloud] (node:199) UnhandledPromiseRejectionWarning: Error: Invalid/Unsupported device type: rt-rpi-300
jenkins-leviathan-v2-template-3208-client-1  |     at new Suite (/usr/app/lib/common/suite.js:158:11)
jenkins-leviathan-v2-template-3208-client-1  |     at /usr/app/lib/common/suiteSubprocess.js:65:16
jenkins-leviathan-v2-template-3208-client-1  |     at Object.<anonymous> (/usr/app/lib/common/suiteSubprocess.js:116:3)
jenkins-leviathan-v2-template-3208-client-1  |     at Module._compile (internal/modules/cjs/loader.js:1085:14)
jenkins-leviathan-v2-template-3208-client-1  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
jenkins-leviathan-v2-template-3208-client-1  |     at Module.load (internal/modules/cjs/loader.js:950:32)
jenkins-leviathan-v2-template-3208-client-1  |     at Function.Module._load (internal/modules/cjs/loader.js:790:12)
jenkins-leviathan-v2-template-3208-client-1  |     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:75:12)
jenkins-leviathan-v2-template-3208-client-1  |     at internal/main/run_main_module.js:17:47
jenkins-leviathan-v2-template-3208-client-1  | (Use `node --trace-warnings ...` to show where the warning was created)
jenkins-leviathan-v2-template-3208-client-1  | [2022-07-24T19:43:37.681Z][90b38fd-cloud] (node:199) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
jenkins-leviathan-v2-template-3208-client-1  | (node:199) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
jenkins-leviathan-v2-template-3208-core-1    | exit
jenkins-leviathan-v2-template-3208-core-1    | Suite exit code is: 0
jenkins-leviathan-v2-template-3208-client-1  | [2022-07-24T19:43:37.698Z][90b38fd-cloud] Test suite has exited with: PASS
```

klutchell, 2022-08-03:
Came here to report the same issue!
```
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:26:58.728Z] INFO: Computing Run Queue
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:26:59.094Z] INFO: [Running Queue] Suites currently in queue: hup,cloud,os
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:26:59.966Z][5e428d6-os] INFO: WORKER URL: https://5e428d69cf3cf9301487f10dc0cf66f8.balena-devices.com/
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:00.367Z][5e428d6-os] Sending to the testbot device, attempt 1...
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:01.074Z][5e428d6-os] Uploading - 10%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:01.782Z][5e428d6-os] Uploading - 21%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:02.488Z][5e428d6-os] Uploading - 32%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:03.196Z][5e428d6-os] Uploading - 43%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:03.899Z][5e428d6-os] Uploading - 55%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:04.604Z][5e428d6-os] Uploading - 66%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:05.307Z][5e428d6-os] Uploading - 77%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:06.011Z][5e428d6-os] Uploading - 88%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:06.724Z][5e428d6-os] Uploading - 99%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:07.924Z][5e428d6-os] Handling artifact: config.json
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:07.925Z][5e428d6-os] Calculating hash
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:07.925Z][5e428d6-os] Sending to the testbot device, attempt 1...
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:07.946Z][5e428d6-os] Uploaded - 100%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:09.924Z][5e428d6-os] Handling artifact: suite
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:09.925Z][5e428d6-os] Calculating hash
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:09.943Z][5e428d6-os] Sending to the testbot device, attempt 1...
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:09.966Z][5e428d6-os] Uploading - 100%
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:12.187Z][5e428d6-os] (node:130) UnhandledPromiseRejectionWarning: Error: Invalid/Unsupported device type: raspberrypi3-unipi-neuron
jenkins-leviathan-v2-template-3434-client-1  |     at new Suite (/usr/app/lib/common/suite.js:158:11)
jenkins-leviathan-v2-template-3434-client-1  |     at /usr/app/lib/common/suiteSubprocess.js:65:16
jenkins-leviathan-v2-template-3434-client-1  |     at Object.<anonymous> (/usr/app/lib/common/suiteSubprocess.js:116:3)
jenkins-leviathan-v2-template-3434-client-1  |     at Module._compile (internal/modules/cjs/loader.js:1085:14)
jenkins-leviathan-v2-template-3434-client-1  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
jenkins-leviathan-v2-template-3434-client-1  |     at Module.load (internal/modules/cjs/loader.js:950:32)
jenkins-leviathan-v2-template-3434-client-1  |     at Function.Module._load (internal/modules/cjs/loader.js:790:12)
jenkins-leviathan-v2-template-3434-client-1  |     at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:75:12)
jenkins-leviathan-v2-template-3434-client-1  |     at internal/main/run_main_module.js:17:47
jenkins-leviathan-v2-template-3434-client-1  | (Use `node --trace-warnings ...` to show where the warning was created)
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:12.188Z][5e428d6-os] (node:130) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
jenkins-leviathan-v2-template-3434-client-1  | (node:130) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
jenkins-leviathan-v2-template-3434-core-1    | exit
jenkins-leviathan-v2-template-3434-core-1    | Suite exit code is: 0
jenkins-leviathan-v2-template-3434-client-1  | [2022-08-03T15:27:12.199Z][5e428d6-os] Test suite has exited with: PASS
```

vipulgupta2048, 2022-08-04:
Yep, this will get resolved with this PR #810

vipulgupta2048, 2022-08-04:
Actually my bad, we are yet to solve the original issue.

rcooke-warwick, 2022-08-04:
@vipulgupta2048 I can fix this one? I think its a matter of changing 2 lines

rcooke-warwick, 2022-08-04:
resolved by https://github.com/balena-os/leviathan/pull/814

closed 2022-08-08 by bulldozer-balena[bot]