balena-os/leviathan #926: Support BoB as a target environment to run Leviathan

vipulgupta2048, 2022-12-29:
Atm, we only support balena-cloud as a target environment on which leviathan runs. 
To unblock, https://github.com/balena-io/e2e/pull/225

We would need to provision a Bob and actually see how leviathan's stack will end up communicating on it and what needs to be made more configurable. 

We later need to understand how Leviathan can be tested against a Bob, staging, and production environments and what changes would that might require.

vipulgupta2048, 2022-12-29:
cc: @rcooke-warwick @alexgg @jakogut  @klutchell Do you have experience running BoB's?

klutchell, 2023-01-02:
@vipulgupta2048 I'm running BoB, and I think everyone is expected to have tried it at some point.

Off the top of my head we would need to expose the URLs used by Leviathan to be configurable. But more importantly, we would need to stop relying on paid features like the proxy.

vipulgupta2048, 2023-01-02:
I haven't tried it or seen it run. I might need to set it up if we can determine priority for this task. 
Not using the proxy might be tough but maybe we can code alternative paths. Let's talk more about this tomorrow see if we can find prioritization for this.

vipulgupta2048, 2023-01-03:
We confirmed we can at least fully start supporting Levithan on staging and scale our way to BoB. 
The solution ideally should work for QEMU and testbot workers both but it could get fairly tricky. We will try our best, but not block the completion of this work.